steptools/NC.js #115: Play/pause button doesn't update properly when two projects are open

Sjbeale, 2016-06-21:
when you play a project it stops the other projects without updating the play/pause button to reflect that it's paused. then if you pause and play another project, pausing the original project that was stopped will update a playing project to display the play button.

Sjbeale, 2016-06-21:
Also if you pause the playing project instead of pausing the original project, the play button will display for the original object but pressing it will do nothing.

funnymanpatrick, 2016-06-24:
Low priority. The supported use cases involve one instance per model so this won't come up

Sam-son, 2016-06-28:
Server is going to be refactored so it has a 1-1 relationship with a file. This nullifies the issue presented here.

closed 2016-06-28 by Sam-son