steptools/NC.js #559: Cannot build

pablo-mayrgundter, 2022-04-03:
Hi,

Just cloned the repo and following the instructions here:

  https://steptools.github.io/NC.js/build.html

After *git clone* and *npm install*, I get the following build error:

```
pablo@top:~/NC.js> uname -a
Darwin MacBook-Air 21.4.0 Darwin Kernel Version 21.4.0 ...; .../RELEASE_ARM64_T8101 arm64
pablo@top:~/NC.js> pablo@top:~/NC.js> npm -v
8.6.0
pablo@top:~/NC.js> npm run make

> NC.js@2.0.0 make
> ./node_modules/.bin/webpack  --mode=development --progress --bail && node src/server/api_server.js -n

/Users/pablo/NC.js/node_modules/webpack-cli/bin/cli.js:281
				throw err;
				^

TypeError: Invalid value used in weak set
    at WeakSet.add (<anonymous>)
    at MiniCssExtractPlugin.apply (/Users/pablo/NC.js/node_modules/mini-css-extract-plugin/dist/index.js:640:18)
    at webpack (/Users/pablo/NC.js/node_modules/webpack/lib/webpack.js:51:13)
    at processOptions (/Users/pablo/NC.js/node_modules/webpack-cli/bin/cli.js:272:16)
    at /Users/pablo/NC.js/node_modules/webpack-cli/bin/cli.js:364:3
    at Object.parse (/Users/pablo/NC.js/node_modules/webpack-cli/node_modules/yargs/yargs.js:576:18)
    at /Users/pablo/NC.js/node_modules/webpack-cli/bin/cli.js:49:8
    at Object.<anonymous> (/Users/pablo/NC.js/node_modules/webpack-cli/bin/cli.js:366:3)

pablo-mayrgundter, 2022-04-03:
Per https://stackoverflow.com/questions/68546117/typeerror-invalid-value-used-in-weak-set-while-build-using-webpack

Downgrading to mini-css-extract-plugin@0.9.0 got me past the first error.

However, it looks like I also hit #558, which is that the dependeny STEPNode is can no longer be compiled by current Nodes and the project itself is ARCHIVED:

https://github.com/steptools/STEPNode

Does that mean NC.js is defunct?

david-loffredo, 2022-04-07:
[I responded to you personally, but posting here for others to see]

Yes, we have not posted an updated STEPNode to NPM in a while, so some versions have aged out.   Some background is in order --

Our vision, going back to 2012, was that a web API and browser client would be easier for others to customize for particular manufacturing situations (which is why it is on github) while we handled the server.  We came at it a few different ways before ending up with this Three.js and Node combo.   The STEPNode package is just a shallow Node wrapper for integration around the STEP Tools commercial C++ libraries for STEP and STEP-NC.

Ultimately browser development didn't turn out to be as productive an environment for third parties as we had hoped.  We switched to .NET and Python as integration environments on the desktop, which manufacturing engineers have an easier time with.

So this project isn't currently active, but isn't abandoned and we expect to revisit it at some point.